afpineda/NuS-NimBLE-Serial #11: std::counting_semaphore is only available from C++20 onwards

ohsix, 2025-02-20:
/home/.../NuS-NimBLE-Serial/src/NuS.hpp:196:3: note: 'std::counting_semaphore' is only available from C++20 onwards
  196 |   std::counting_semaphore<1> peerConnected{std::counting_semaphore<1>(0)};
      |   ^~~

the "official" / de-facto c++ spec supported by arduino is c++11*

*: long story, short: supporting combinations of arbitrary versions is hard, and impossible in some cases

pixel3design-hub, 2025-02-28:
Hi, i got issue compiling in Platformio with version 3.3.3 of the library:

```
In file included from .pio/libdeps/esp32-c3-devkitm-1/NuS-NimBLE-Serial/src/NuSerial.hpp:16,
                 from include/restapi.cpp:6,
                 from src/main.cpp:6:
.pio/libdeps/esp32-c3-devkitm-1/NuS-NimBLE-Serial/src/NuStream.hpp:111:65: error: use of deleted function 'cyan::counting_semaphore<least_max_value>::counting_semaphore(const cyan::counting_semaphore<least_max_value>&) [with int least_max_value = 1]'
     counting_semaphore<1> dataAvailable{counting_semaphore<1>(0)};
                                                                 ^
```
I'm not entirely sure, but I think that's a related issue...

pixel3design-hub, 2025-03-01:
Seem that the issue with semaphores is fixed.
Now just Platformio won't accept the newer version and refuses to install the 3.3.4 version, but that's for another thread.
Thanks and cheers!